ASCOMInitiative/ASCOMRemote #3: Call /api/v1/Camera/0/ImageArray make ASCOMRemote freeze.

elendil-software, 2018-09-20:
Hi,

When I try to call /api/v1/Camera/0/ImageArray the app freeze (Not Responding). I tried with the versions 1.0.1 and 1.1.
I tried with my QSI Camera and with the simulator.

Here is the log : 

> 20:27:02.572   0      0      0         WebRequestCallback     Thread 4 - Request received. Is thread pool thread: True. Is background thread: True.
20:27:02.573   0      0      0         WebRequestCallback     Thread 4 - Setting up new call back to wait for next request 
20:27:02.573   0      0      0         WebRequestCallback     Thread 4 - Processing received message.
20:27:02.595   0      0      10        RequestReceived        GET URL: /api/v1/Camera/0/ImageArray, Thread: 4
20:27:02.595   0      0      10        RequestReceived        Header Postman-Token = 784a5235-5ace-444f-8816-8a82ab6c2b68
20:27:02.595   0      0      10        RequestReceived        Header Cache-Control = no-cache
20:27:02.595   0      0      10        RequestReceived        Header Connection = keep-alive
20:27:02.595   0      0      10        RequestReceived        Header Content-Type = application/x-www-form-urlencoded
20:27:02.595   0      0      10        RequestReceived        Header Accept = application/json
20:27:02.595   0      0      10        RequestReceived        Header Accept-Encoding = gzip, deflate
20:27:02.596   0      0      10        RequestReceived        Header Host = 192.168.1.101:11111
20:27:02.596   0      0      10        RequestReceived        Header User-Agent = PostmanRuntime/7.3.0
20:27:15.441   0      0      10        ImageArray             ImageArray Rank: 2
20:27:15.441   0      0      10        ImageArray             ImageArray Rank: 2, Length: 8328304
20:27:15.460   0      0      10        ImageArray             Starting array serialisation
20:27:17.790   0      0      10        ImageArray             Completed array serialisation, Length: 33339081
20:27:17.790   0      0      10        ImageArray             OK - no exception. Thread: 4, Json: {"Type":2,"Rank":2,"Value":[[365,368,400,365,384,386,392,[...],
20:27:17.848   0      0      10        ImageArray             Completed Encoding.GetBytes, array length: 33339081

Regards

Julien

Peter-Simpson, 2018-09-20:
Hi Julien,

To cross check this I've just run Conform through a Remote Client camera driver to an instance of the Remote Server and it passed OK including returning  ImageArray ImageArrayVariant so I'm wondering whether this is in some way related to the way the server is interacting with your network. 

Please can you help me with some more information.

1. How you are sending commands to the Remote Server, e.g. a browser or the Camera remote client driver?
1. Do you have Debug logging enabled? If not please can you enable this in the Setup GUI.
1. Please can you post the entire session log rather than just an extract
1. Are you running the client and server on the same PC or on different PCs?
1. If different PCs how are they networked together?
1. Please can you post a screenshot (Alt/PrtScn) of the Camera Simulator configuration GUI.

Thanks, Peter

elendil-software, 2018-09-21:
I think I have found why it didn't work. I was trying with Postman (https://www.getpostman.com/). And it looks like it limit the size of the returned data. I think we can close this issue.

elendil-software, 2018-09-23:
The problem is there again, even with TheSkyX
Here are logs and screenshot of configuration dialogs : 
[logs.zip](https://github.com/ASCOMInitiative/ASCOMRemote/files/2408770/logs.zip)

elendil-software, 2018-09-24:
Hi,

Yes, I'm from Switzerland.

Still an error. 
[Logs 2018-09-24.zip](https://github.com/ASCOMInitiative/ASCOMRemote/files/2412517/Logs.2018-09-24.zip)

Regards

Julien

Peter-Simpson, 2018-09-24:
Hi Julien,

Thanks for the logs, I'll go through them in detail to see if I can figure out what is going on and will set up a Swiss locale test environment. Its possible that this is happening because the drivers are now running in separate threads, this is a new feature that I recently added. 

Please can you test whether the problem still exists when the drivers run on the main thread by unchecking the "Run drivers in separate threads" checkbox in the server configuration screen and retesting.

I couldn't see the Diagnostics log in your last ZIP file, I'd appreciate it if you can you also post this.

Many thanks, Peter

[image]

Peter-Simpson, 2018-09-24:
Hi Julien,

One more thought, do you have the "Log Responses" checkbox checked on the main server screen? If so, please can you try a test with it unchecked, it may be that the list box is unhappy with the characters being logged from your Swiss locale.

Regards, Peter

[image]

Peter-Simpson, 2018-09-24:
Hi Julien,

I've been able to reproduce the issue here now, it occurs only when "Log Responses" is enabled, I'll check this through tomorrow and let you have a fixed version. This was working OK but has clearly broken at some point, as you can tell, I don't usually log the JSON responses!

Regards, Peter

elendil-software, 2018-09-25:
Hi,

Thanks.

I confirm, it works when Log Responses is disabled

Regards
Julien

Peter-Simpson, 2018-09-25:
Hi Julien,

That's good news! I've attached a new version that should now work with Log Responses enabled. 

The cause of the issue was that the 1.8Mb ImageArray JSON response was more than the status text box could handle. I've now truncated displayed JSON responses to 500 characters and all seems to be well.

Please let me know how you get on.

Regards, Peter

[ASCOMRemote(1.1.6842.25892)setup.exe.zip](https://github.com/ASCOMInitiative/ASCOMRemote/files/2416614/ASCOMRemote.1.1.6842.25892.setup.exe.zip)

elendil-software, 2018-09-26:
Perfect ! Thanks

Julien

Peter-Simpson, 2018-09-26:
Great good news Julien!

Regards, Peter

closed 2018-09-26 by Peter-Simpson