ASCOMInitiative/ASCOMRemote #4: Allow for custom park and home positions

dyfiastro, 2019-03-17:
I am currently trying to get ascom remote working with multiple EQMOD controlled mounts.
These all have custom home positions set to allow for the roof to roll on and off.

Without the ability to select a custom home position to park to using ascom remote I still currently need to log in to the remote PC to set the custom home position.

Peter-Simpson, 2019-03-17:
Hi,

Are you saying that Telescope.SetPark is not working in ASCOM Remote?

Regards Peter

dyfiastro, 2019-03-17:
No it seems to park to the standard park position (Scope up, weights down).
However I have separate park positions that can be called to allow the roof to be rolled on and off.
I cannot see a way to call these as a separate option.

Peter-Simpson, 2019-03-17:
Would the Action and SupportedActions methods help? 

⁣Sent from BlueMail ​

dyfiastro, 2019-03-17:
I am not sure to be honest but its worth trying

Peter-Simpson, 2019-05-24:
Closed because this isn't an issue with ASCOM Remote and no further update from dyfiastro.

closed 2019-05-24 by Peter-Simpson