ASCOMInitiative/ASCOMRemote #5: EquatorialCoordinateType.equTopocentric doesn't exist, perhaps  EquatorialCoordinateType.equLocalTopocentric was intended?

mitchmitchell, 2019-03-18:
Line 3400 of ServerForm.cs function ReturnEquatorialSystem attempts to set deviceResponse to EquatorialCoordinateType.equTopocentric which doesn't exist in the enum definition for EquatorialCoordinateType -- possibly what is meant is:  equLocalTopocentric?

Peter-Simpson, 2019-03-18:
Hi Mitch,

equLocalTopocentric has been deprecated in favour of equTopocentric because topocentric already implies "local". Using the term "LocalTopocentric" is like saying "local local position".

The change is documented here: https://ascom-standards.org/Help/Platform/html/T_ASCOM_DeviceInterface_EquatorialCoordinateType.htm 

Regards Peter 

⁣Sent from BlueMail ​

mitchmitchell, 2019-03-18:
Nonetheless,  the solution refused to compile when cloned from the repository because the enumeration did not define equTopocentric.  So I guess the definition needs to be added to the enumeration declaration.

Sent via the Samsung Galaxy Note8, an AT&T 4G LTE smartphone

Peter-Simpson, 2019-04-02:
Hi Mitch,

equTopocentric was added to the enumeration in June 2018, are you using the latest Platform Developer Components release?

Regards, Peter

Peter-Simpson, 2019-05-24:
Closed on the basis of no feedback as to whether using the current developer tools fixed the issue.

closed 2019-05-24 by Peter-Simpson