ASCOMInitiative/ASCOMRemote #6: Exception after updated drivers

mgrant9999, 2019-04-02:
The following occurred after updating the drivers on my machine to the latest Atik drivers v4.2.2.9.  The exception pops up after hitting setup on the remote server UI.

Main                      Thread exception: System.NullReferenceException: Object reference not set to an instance of an object.
   at ASCOM.Remote.ServedDevice.CmbDevice_SelectedIndexChanged(Object sender, EventArgs e) in J:\ASCOMRemote\Remote Server\ServedDevice.cs:line 277
   at System.Windows.Forms.ComboBox.OnSelectedIndexChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.set_SelectedIndex(Int32 value)
   at ASCOM.Remote.ServedDevice.set_ProgID(String value) in J:\ASCOMRemote\Remote Server\ServedDevice.cs:line 170
   at ASCOM.Remote.SetupForm.Form_Load(Object sender, EventArgs e) in J:\ASCOMRemote\Remote Server\SetupForm.cs:line 123
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Peter-Simpson, 2019-04-02:
I've recently added some code to help identify initialisation issues. Please can you 

- Install this version: [ASCOM Remote](https://download.ascom-standards.org/beta/ASCOMRemote(1.5.7011.26148)setup.exe) 
- Check the Remote Server Logging and Debug Logging setup checkboxes
- Exit the Remote Server
- Restart the remote server
- Enter Setup (to cause the failure)
- Zip up the current log file directory, Documents\ASCOM\Logs 2019-04-XX,
- Post the zip file here

Many thanks, Peter

mgrant9999, 2019-04-03:
[Logs 2019-04-02.zip](https://github.com/ASCOMInitiative/ASCOMRemote/files/3036338/Logs.2019-04-02.zip)
Peter, Please let me know if you need more.
Thanks!

Peter-Simpson, 2019-04-03:
From the logs it may be that the camera driver ASCOM.Atik2.Camera is not correctly registered for COM and that this is making ASCOM Remote fall over. I've added some extra logging to test this hypothesis in a new version here: [ASCOM Remote](https://download.ascom-standards.org/beta/ASCOMRemote(1.5.7032.20407)setup.exe).

Please can you install it and run the tests as above to see if the behaviour is different this time.

Please can you also run the ASCOM Diagnostics program in the start menu folder Start\ASCOM Platform 6 before zipping up the logs folder.

It would also be interesting to know whether Diagnostics can load the driver through its "Choose Device" menu item.

Regards, Peter

mgrant9999, 2019-04-04:
[Logs 2019-04-03.zip](https://github.com/ASCOMInitiative/ASCOMRemote/files/3041205/Logs.2019-04-03.zip)
Peter,
Results have been uploaded.  Hopefully, this will shed light on the issues I'm experiencing. Can't help but thinking you're right, in that this was a dirty re-install of updated vendor drivers causing exceptions.   I was able to get past the setup blocker with the new version.   I've also enclosed 2 screenshots showing the selections available from 'choose device'.   I much appreciate the help, this development will greatly benefit the ASCOM community.

Peter-Simpson, 2019-04-04:
I'm glad you were able to get further this time. The Remote Server log shows that it is not possible to create a driver with your configured ProgID "ASCOM.Atik2.Camera" and  Diagnostics doesn't show any evidence of a device with this ProgID being installed on your PC.

However, it does show evidence of these two Atik camera entries:

```
19:51:30.894 RegistrySecurityRec       OK - SubKey \Camera Drivers\ASCOM.AtikCameras.Camera does have full registry access rights for BUILTIN\Users
19:51:30.918 RegistrySecurityRec       OK - SubKey \Camera Drivers\ASCOM.AtikCameras2.Camera does have full registry access rights for BUILTIN\Users
```

Your first screenshot shows that you are getting the correct ASCOM.AtikCameras2.Camera ProgID when selecting the second Atik camera through the Chooser and I'm seeing the expected results in the device connection tester.

My hypothesis is that Atik changed the ProgID of the second camera driver in their latest installer and that ASCOM.Atik2.Camera is left over from an earlier install and should have been cleaned up by the installer but was missed for some reason. This would explain why your configuration worked OK before the Atik upgrade and failed afterwards.

I suspect that, if you select the second Atik camera again in the Remote Server configuration, you will get the correct ASCOM.AtikCameras2.Camera ProgID and that the camera will work as expected.

Please can you try this and let me know the outcome?

Regards, Peter

mgrant9999, 2019-04-07:
Peter,
Apologies for the delay in responding, as I was attending NEAIC/NEAF here on the east coast. I was able to successfully set up both cameras, my mount and filter wheel.  Ran an extensive series of tests and encountered no problems.  The only issue I did encounter was sorting out the proper firewall settings to apply.  Having done that, the testing went smoothly. I'm convinced the original issues were all due to improper rollback of the installed drivers.  Thanks for all of your help in sorting this out!  If you need any additional field testing etc. please let me know.
Regards,
Mike

Peter-Simpson, 2019-04-08:
Regards, Peter

Peter-Simpson, 2019-05-24:
Closed - issue resolved.

closed 2019-05-24 by Peter-Simpson