ASCOMInitiative/ASCOMRemote #7: Unexpected Error in ServerForm

T0T4R4, 2019-05-06:
```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Collections.Generic.KeyNotFoundException: The given key was not present in the dictionary.
   at System.Collections.Concurrent.ConcurrentDictionary`2.get_Item(TKey key)
   at ASCOM.Remote.ServerForm.CreateInstance(KeyValuePair`2 configuredDevice) in J:\ASCOMRemote\Remote Server\ServerForm.cs:line 613
   at ASCOM.Remote.DriverHostForm.DriverHostForm_Load(Object sender, EventArgs e) in J:\ASCOMRemote\Remote Server\DriverHostForm.cs:line 51
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3062.0 built by: NET472REL1
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
ASCOM.RemoteServer
    Assembly Version: 1.5.7008.22516
    Win32 Version: 1.5.7008.22516
    CodeBase: file:///C:/Program%20Files%20(x86)/ASCOM/Remote/ASCOM.RemoteServer.exe
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3062.0 built by: NET472REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3062.0 built by: NET472REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3062.0 built by: NET472REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
ASCOM.Utilities
    Assembly Version: 6.0.0.0
    Win32 Version: 6.3.0.2831
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/ASCOM.Utilities/6.0.0.0__565de79
```

Peter-Simpson, 2019-05-07:
Hi 

I can't respond helpfully to this since you have provided no context whatsoever, all I can determine from this post is that you are running ASCOM Remote Server.

I have made some recent improvements to resilience and logging. Please can you install this latest version and check the "Write log files" and "Enable debug logging" check boxes on the Setup dialogue to see if this version generates better diagnostics. [ASCOM Remote](https://download.ascom-standards.org/beta/ASCOMRemote(1.5.7061.24069)setup.exe)

Please post the resultant log file, which will be in your Documents\ASCOM\Logs 2019-05-XX folder together with a list of devices that you have configured and details of what triggers this crash

Regards, Peter

T0T4R4, 2019-05-08:
Hi Peter, 

Sorry for not providing more context. I have a dome simulator attached and a telescope as well (via PWI). 

I will do as indicated and post the logs, thanks.

Peter-Simpson, 2019-06-10:
Closing this after 1 month due to lack of further activity, on the assumption that the revised ASCOM Remote version provided enough information to resolve the problem.

Regards, Peter.

closed 2019-06-10 by Peter-Simpson