ASCOMInitiative/ASCOMRemote #9: How to interface with the management API

zwik, 2019-05-08:
I just installed the latest version of ASCOMRemote and am able to interface with it. According to the documentation I should find a webinterface on /setup, or I should be able to get a list of available API on /management/apiversions. Both of them return a HTML page with "not found". I'm probably doing something wrong, but what? Can someone help me out?

Thanks!

Peter-Simpson, 2019-05-08:
Hi Marcel,

 

We only finalised the management API recently and I haven’t had time to retrofit it to ASCOM Remote yet. I will put it on the request list, it certainly should be there!

 

Regards, Peter

zwik, 2019-05-09:
Thanks Peter,

I just noticed in the settings that the management page is disabled. Is it an actual web page then or should I be able to configure/read everything from a JSON API? I would love a full JSON where configuration is only done with an API.

Peter-Simpson, 2019-05-13:
Hi Marcel,

 

It’s deactivated because the current management API doesn’t comply with the standard, I’ll enable it again as soon as I’ve added the new capability.

 

Regards, Peter

Peter-Simpson, 2020-01-28:
I'm closing this issue because the management interface and an HTML setup page have now been implemented in accordance with the Alpaca API. These will be released alongside the forthcoming Platform 6.5 release in the next one to two months.

closed 2020-01-28 by Peter-Simpson