ASCOMInitiative/ASCOMRemote #13: Cannot create Camera device when settings are misconfigured.

jwellman80, 2020-04-30:
Attempting to create the `ASCOM.Remote1.Camera` device and open the SettingsDialog via the built in ASCOM.DriverAccess methods results in a `"Communications exception: Unable to connect to the remote server - Error"` being thrown.  

Steps to reproduce: 
1 - Stop the server or have the IP of the ASCOM.Remote1.Camera misconfigured
2 - Create the ASCOM.Remote1.Camera device using the ASCOM.DriverAccess.Camera method
3 - Receive Exception.  

I presume that something is happening in the constructor of the ASCOM.Remote1.Camera device that is attempting to reach out to the Server and the exception is being thrown.  Unfortunately this means that the SetupDialog cannot be called to correct the IP.   It could be that this is happening in the ASCOM Platform but it seemed more likely to be in the ASCOMRemote...but that's just a hunch.

Using 
 - ASCOM Platform 6.5 RC2
 - ASCOMRemote v1.6.7424.28304

Peter-Simpson, 2020-04-30:
Hi,

What is happening is that DriverAccess needs to know the driver's interface version so that it can configure its behaviour accordingly. DriverAccess always presents the latest interface version but has to adapt its behaviour when it encounters drivers with earlier interface versions.

In this scenario you could catch the DriverException exception and use the static Camera.Choose("ASCOM.Remote1.Camera") method to display a Chooser dialogue. The Setup can then be reconfigured from the Chooser's properties button without connecting to the driver. I've copied some code below.

Unfortunately the code exposed an issue in the revised Platform Chooser code, which throws a threading exception. Please can you try this new version: [Platform 6.5 RC2](https://download.ascom-standards.org/beta/ASCOMPlatform65.exe), which should display the Chooser correctly. 

Regards, Peter

>try
>{
>   Camera camera = new Camera("ASCOM.Remote1.Camera");
> 
>   MessageBox.Show($"Description: {camera.Description}");
>   camera.Dispose();
> 
>}
>catch(DriverException ex)
>{
>   MessageBox.Show(ex.ToString());
> 
>   // Display a Chooser 
>   Camera.Choose("ASCOM.Remote1.Camera");
> 
>}

Peter-Simpson, 2020-05-01:
Hi,

I've just reworked DriverAccess so that querying the driver's interface version is done lazily, only when required. This means that you original code construct should work as expected.

The revised Platform is here: [Platform 6.5 RC3](https://download.ascom-standards.org/beta/ASCOMPlatform65.exe)
The revised Developer Components are here: [Platform 6.5 Developer Components RC3](https://download.ascom-standards.org/beta/ASCOMPlatform65Developer.exe)

Please let me know if this version fixes the issue.

Thanks, Peter

Peter-Simpson, 2022-01-10:
Closing because the fix has been delivered.

closed 2022-01-10 by Peter-Simpson