ASCOMInitiative/ASCOMRemote #14: Chunked encoding upload is not supported on the HTTP/1.0 protocol.

mikeread2018, 2020-05-06:
Hi,

I have just installed 1.6.7424.28304 and get the following when taking an image with an ASI533. Anything I can do to resolve this?

Running on Windows 10 x64 Pro

16:16:51.329 Main                      Unhandled exception: System.Net.ProtocolViolationException: Chunked encoding upload is not supported on the HTTP/1.0 protocol.
   at System.Net.HttpListenerResponse.set_EntitySendFormat(EntitySendFormat value)
   at ASCOM.Remote.ServerForm.ReturnImageArrayBase64(RequestData requestData) in J:\ASCOMRemote\Remote Server\ServerForm.cs:line 4946
   at ASCOM.Remote.ServerForm.ProcessDriverCommand(RequestData requestData) in J:\ASCOMRemote\Remote Server\ServerForm.cs:line 2723
   at ASCOM.Remote.DriverHostForm.ProcessCommand(Object requestData) in J:\ASCOMRemote\Remote Server\DriverHostForm.cs:line 97
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)

Peter-Simpson, 2020-05-06:
Hi Mike,

Please can you confirm what client application and client driver you are using and also describe the network topology between the client and the server. 

This issue isn't one that I've seen before in almost 2 years of use so there may be something about your particular environment that is triggering the behaviour or it could be something wrong in Platform 6.5.

Thanks, Peter

mikeread2018, 2020-05-06:
Hi Peter,

Thanks for coming back.

Client Application:

CCDciel
Version beta 0.9.70-2181-efefc61 2020/04/09 09:51:58
Compiled with:
Lazarus 2.0.6.0 Free Pascal 3.0.4 Win32-i386-win32

The “server” is running on a Dell 3070 I3, 4Gb RAM. Windows 10 Pro. Connected over WiFi (2Ghz).  The client PC is a Dell XPS 15 I7 9th Gen. Also connected over WiFi.

The Camera ASI533 is connected USB3.

Happy to run any diags etc…

Kind regards

Mike

mikeread2018, 2020-05-06:
Hi Peter, Also just disconnected from the WiFi and Cat6 to a switch - still the same issue

Peter-Simpson, 2020-05-06:
Hi Mike,

Thanks for testing, I think I now know what is going on. CdC uses its own built-in Alpaca client rather than either an ASCOM Remote Client or a Dynamic Client that comes with Platform 6.5. The clients that I've produced all use HTTP/1.1 and support chunked encoding, so the Remote Server was hard wired to enable chunked encoding.

It seems that CdC is using HTTP/1.0, which doesn't support chunked encoding, and consequently the Remote Server was complaining when it tried to enable chunking support.

I've now made the Remote Server only enable chunking when an HTTP/1.1 connection is in use and so it should now work correctly with CdC. please can you give this version a go?

[ASCOM Remote Installer](https://download.ascom-standards.org/beta/ASCOMRemote(1.6.7431.32351)setup.exe)

Thanks, Peter

mikeread2018, 2020-05-06:
Hi Peter,

I’m still getting this:

20:16:19.769 Main                      Unhandled exception: System.Net.ProtocolViolationException: Chunked encoding upload is not supported on the HTTP/1.0 protocol.
   at System.Net.HttpListenerResponse.set_EntitySendFormat(EntitySendFormat value)
   at ASCOM.Remote.ServerForm.ReturnImageArrayBase64(RequestData requestData) in J:\ASCOMRemote\Remote Server\ServerForm.cs:line 4949
   at ASCOM.Remote.ServerForm.ProcessDriverCommand(RequestData requestData) in J:\ASCOMRemote\Remote Server\ServerForm.cs:line 2723
   at ASCOM.Remote.DriverHostForm.ProcessCommand(Object requestData) in J:\ASCOMRemote\Remote Server\DriverHostForm.cs:line 97
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)

Peter-Simpson, 2020-05-06:
Hi Mike,

That's unexpected, to help me track this down, please can you:
- Install this version, it will still fail but is a debug build to ensure that the reported line numbers are correct. [https://download.ascom-standards.org/beta/ASCOMRemote(1.6.7431.37391)setup.exe](https://download.ascom-standards.org/beta/ASCOMRemote(1.6.7431.37391)setup.exe) 
- enable logging and debug logging on the Remote Server setup
- reproduce the issue
- attach the entire remote server log file so I can see what is happening. It will be in your Documents\ASCOM\Logs 2020-05-06 folder.

Many thanks, Peter

Peter-Simpson, 2020-05-07:
Hi Mike,

I've had another go at this and added some further protection; hopefully this version will work. Please let me know how you get on.

[https://download.ascom-standards.org/beta/ASCOMRemote(1.6.7432.18168)setup.exe](https://download.ascom-standards.org/beta/ASCOMRemote(1.6.7432.18168)setup.exe)

Thanks, Peter

mikeread2018, 2020-05-07:
Thanks Peter,

I’ll give a test later this afternoon.

Please excuse any predictive text and typos!

mikeread2018, 2020-05-07:
Hi Peter,

Thanks for this, that has worked. I can now get images from CCDCiel using the remote server.

I’ll continue to test and raise any other issues I find.

Once again, thank you for your swift response.

Peter-Simpson, 2020-05-07:
Hi Mike,

Excellent news, I'm glad that fixed the issue. It will help me understand what is going on with CdC's client if you can still send me a Remote Server log file with logging and debug logging enabled as described above.

Many thanks, Peter

Peter-Simpson, 2020-05-11:
I'm closing this issue because it has been resolved.

closed 2020-05-11 by Peter-Simpson