ASCOMInitiative/ASCOMRemote #15: Is there supposed to be a client option for Cover?

linuxkidd, 2020-06-17:
Installed ASCOM 6.5RC2 and ASCOM Remote 1.6.7424.28304.  I see in the 'Remote Server' lists a Cover device, but the 'Remote Client Configuration' dialog doesn't have a Cover option. 

Am I missing something?

linuxkidd, 2020-06-18:
@Peter-Simpson I don't think your update came through the way you intended... :)

Peter-Simpson, 2020-06-19:
Hi Michael,

Very sorry, I really have no idea how this happened, nor the same on #16!

Platform 6.5 has a new feature called "Dynamic Clients" that do much the same job as the Remote Clients". The Dynamic Clients can be created on demand through the Chooser's new Alpaca menu. 

Rather than maintain these two two different approaches, we've decided to stop adding new features to the Remote Clients and just to concentrate on developing the Dynamic Clients in future. CoverCalibrators are a new device type and this is why support only appears in the Dynamic Clients.

Here is a link showing how to use Dynamic Clients, hopefully this is what you are looking for. https://ascom-standards.org/platform65rchelp/html/e3870a2f-582a-4ab4-b37f-e9b1c37a2030.htm 

Best wishes, Peter

linuxkidd, 2020-06-20:
Hi Peter, thanks for these details.  Makes complete sense.  I consider this resolved.

closed 2020-06-20 by linuxkidd