ASCOMInitiative/ASCOMRemote #16: Looking for documentation for CoverCalibrator api endpoints

linuxkidd, 2020-06-17:
Is there documentation for CoverCalibrator api endpoint?   I haven't located it in `ASCOM Alpaca API Reference.pdf` file or the https://ascom-standards.org/api/ site.

Thanks!

linuxkidd, 2020-06-18:
@Peter-Simpson I don't think your update came through the way you intended... :)

Peter-Simpson, 2020-06-19:
Hi Michael,

As #15, I really don't know how my empty post came about, sorry!

We haven't finished the online documentation for the CoverCalibrator Alpaca API yet, there isn't an issue of any kind, it will follow the same principles as the other APIs, the timing is just down to workload prioritisation. 

If you examine how similar methods are implemented in other interfaces, I think you will be able to deduce pretty much what the interface will be.

We hope to publish Platform 6.5 (including the Alpaca API spec) next week.

Best wishes, Peter

linuxkidd, 2020-06-20:
Sounds great Peter, thanks for the update! :)

jamessynge, 2021-07-11:
This can probably be closed now.

Peter-Simpson, 2021-07-11:
The API has been published, so I'm closing this issue as suggested.

Best wishes, Peter

closed 2021-07-11 by Peter-Simpson