ASCOMInitiative/ASCOMRemote #18: Minor API doc string typos

lukastoenneMS, 2020-10-21:
Nitpicking:
* Az/RA/Dec says "telescope" instead of "mount", and lower case: /telescope​/{device_number}​/altitude: Returns the mount's Altitude above the horizon.
* "Azimuth": ​/telescope​/{device_number}​/azimuth: Returns the telescope's aperture.

Peter-Simpson, 2020-10-26:
Hi Lukas,

Thanks for reporting this, in which document did you spot these errors?

Thanks, Peter

lukastoenneMS, 2020-10-26:
Hi Peter,

Found these on the API doc page, looks like in the repo they are here:

https://github.com/ASCOMInitiative/ASCOMRemote/blob/84b2806c4ac5a8e962c806d1751852f49809592e/Swagger/ASCOMInitiative_ASCOMDeviceAPI_1.0.1-oas3_swagger.yaml#L5471

https://github.com/ASCOMInitiative/ASCOMRemote/blob/84b2806c4ac5a8e962c806d1751852f49809592e/Swagger/ASCOMInitiative_ASCOMDeviceAPI_1.0.1-oas3_swagger.yaml#L5625

Peter-Simpson, 2020-10-28:
Hi Lukas,

Many thanks, I've corrected the typos and the revised versions are online. You may need to CTRL/F5 refresh the page to overwrite the earlier version that may be cached in your browser.

Best wishes, Peter

commit 4082c27e (2020-10-28): Correct typos in Alpaca device API definition - Fixes #18.
closed 2020-10-28 by Peter-Simpson