ASCOMInitiative/ASCOMRemote #19: Bug on 'error' received by Remote Server

skybadger, 2020-10-26:
Setup : 
ESP8266 Dome ALPACA device returns a float value to a Dome get interface version call. 
Remote server instance gets upset at unexpected float when looking for an int.
But notice the unexpected reference to Image Array reference Type ? 
Logs from remote Server logging.

21:44:42.411 DriverInfo 34657          ASCOM Remote Driver v1.6.7500.21910 - REMOTE DEVICE: Skybadger.ESPDome RESTful native device. 
21:44:42.415 DriverVersion 34657       Client Txn ID: 2053, Sending command to remote server
21:44:42.566 DriverVersion 34657       Response Status: 'OK', Response: {"Value":"0.9","ClientID":34657,"ClientTransactionID":2053,"ServerTransactionID":269,"Method":"DriverVersion","ErrorNumber":0,"ErrorMessage":""}
21:44:42.566 DriverVersion 34657       Client Txn ID: 2053, Server Txn ID: 269, Value: 0.9
21:44:42.566 CallWasSuccessful         DriverException == null: True, ErrorMessage: , ErrorNumber: 0x00000000
21:44:42.566 CallWasSuccessful         Returning True
21:44:42.566                           
21:44:42.566 DriverVersion 34657       0.9
21:44:42.572 InterfaceVersion 34657    Client Txn ID: 2054, Sending command to remote server
21:44:42.739 InterfaceVersion 34657    Response Status: 'OK', Response: {"Value":"0.9","ClientID":34657,"ClientTransactionID":2054,"ServerTransactionID":270,"Method":"InterfaceVersion","ErrorNumber":0,"ErrorMessage":""}
21:44:42.739 InterfaceVersion 34657    Int16 Could not convert string to integer: 0.9. Path 'Value', line 1, position 14.
21:44:42.740 InterfaceVersion 34657    Exception: Newtonsoft.Json.JsonReaderException: Could not convert string to integer: 0.9. Path 'Value', line 1, position 14.
   at Newtonsoft.Json.JsonReader.ReadInt32String(String s)
   at Newtonsoft.Json.JsonTextReader.FinishReadQuotedNumber(ReadType readType)
   at Newtonsoft.Json.JsonTextReader.ReadNumberValue(ReadType readType)
   at Newtonsoft.Json.JsonTextReader.ReadAsInt32()
   at Newtonsoft.Json.JsonReader.ReadForType(JsonContract contract, Boolean hasConverter)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
   at Newtonsoft.Json.JsonSerializer.DeserializeInternal(JsonReader reader, Type objectType)
   at Newtonsoft.Json.JsonConvert.DeserializeObject(String value, Type type, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value, JsonSerializerSettings settings)
   at Newtonsoft.Json.JsonConvert.DeserializeObject[T](String value)
   at ASCOM.Remote.RemoteClientDriver.SendToRemoteDriver[T](UInt32 clientNumber, RestClient client, String URIBase, TraceLoggerPlus TL, String method, Dictionary`2 Parameters, Method HttpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression) in J:\ASCOMRemote\Remote Client Local Server\RemoteClientDriver.cs:line 614
21:44:42.748 SupportedActions 34657    Client Txn ID: 2055, Sending command to remote server
21:44:43.142 SupportedActions 34657    Response Status: 'Bad Request', Response: {"Value":0,"ClientID":34657,"ClientTransactionID":2055,"ServerTransactionID":271,"Method":"HandlerNotFound","ErrorNumber":1035,"ErrorMessage":"No REST handler found for argument - check ASCOM Dome v2 specification"}
21:44:43.142 SupportedActions Error 34 RestRequest response status: Completed, HTTP response code: BadRequest, HTTP response description: Bad Request
21:44:43.142 SupportedActions 34657    List`1 Error calling method: SupportedActions, HTTP Completion Status: Completed, Error Message:
{"Value":0,"ClientID":34657,"ClientTransactionID":2055,"ServerTransactionID":271,"Method":"HandlerNotFound","ErrorNumber":1035,"ErrorMessage":"No REST handler found for argument - check ASCOM Dome v2 specification"}
21:44:43.143 SupportedActions 34657    Exception: ASCOM.DriverException: Error calling method: SupportedActions, HTTP Completion Status: Completed, Error Message:
{"Value":0,"ClientID":34657,"ClientTransactionID":2055,"ServerTransactionID":271,"Method":"HandlerNotFound","ErrorNumber":1035,"ErrorMessage":"No REST handler found for argument - check ASCOM Dome v2 specification"}
   at ASCOM.Remote.RemoteClientDriver.SendToRemoteDriver[T](UInt32 clientNumber, RestClient client, String URIBase, TraceLoggerPlus TL, String method, Dictionary`2 Parameters, Method HttpMethod, ImageArrayTransferType imageArrayTransferType, ImageArrayCompression imageArrayCompression) in J:\ASCOMRemote\Remote Client Local Server\RemoteClientDriver.cs:line 1121
21:44:43.150 ShutterStatus 34657       Client Txn ID: 2056, Sending command to remote server
21:44:43.296 ShutterStatus 34657       Response Status: 'OK', Response: {"Value":4,"ClientID":34657,"ClientTransactionID":2056,"ServerTransactionID":272,"Method":"ShutterStatus","ErrorNumber":0,"ErrorMessage":""}
21:44:43.296 ShutterStatus 34657       Client Txn ID: 2056, Server Txn ID: 272, Value: 4
21:44:43.296 CallWasSuccessful         DriverException == null: True, ErrorMessage: , ErrorNumber: 0x00000000
21:44:43.296 CallWasSuccessful         Returning True

Peter-Simpson, 2020-10-28:
Hi,

This looks OK to me, the references to image array variables are input parameters to the SendToRemoteDriver[T] method and any supplied values are ignored for when dealing with no image types.

BTW, the ASCOM InterfaceVersion relates to the integer version number of the ASCOM interface that you are implementing rather than your own release version number.

best wishes, Petger

skybadger, 2020-10-28:
Right, thanks for following up on that. 

----- Reply message -----

closed 2020-10-29 by Peter-Simpson