ASCOMInitiative/ASCOMRemote #24: ASCOM Alpaca Switch - 'Value=' is not added to 'setswitchvalue' request URL.

RomanGmprln, 2021-04-12:
Hello,

I am trying to control a ASCOM Alpaca Switch device. The device supports multiple switchvalues: [0.0, 1.0, 2.0]

‌

On setting a new switchvalue (eg. to 2.0 ) in N.I.N.A., the http request URL lacks the “Value=2.0“. (see log line 58))

/api/v1/switch/0/setswitchvaluewhere I am expecting:

/api/v1/switch/0/setswitchvalue?Value=2.0&ClientTransactionID=XXXX&ClientID=XXXX&ID=0

‌
Looking at the log, I see the other requests are correct and do also contain a ClientTransaction and ClientID. Where the ‘setswitchvalue’ PUT request does not provide a 'Value’. The ClientTransactionID is internally incremented but not added to the request URL. (compare log line 55, 58 and 61).
‌

Trying it out on: https://ascom-standards.org/api/#/Switch%20Specific%20Methods/put_switch__device_number__setswitchvalue

it seems to me, that the Alpaca API reference does not provide the correct request string. Where the exemplary curl string has the Value

curl -X PUT "https://virtserver.swaggerhub.com/ASCOMInitiative/api/v1/switch/0/setswitchvalue" -H "accept: application/json" -H "Content-Type: application/x-www-form-urlencoded" -d "Id=&Value=2.0&ClientID=0&ClientTransactionID=1234"
the generated request URL lacks the Value and IDs:

https://virtserver.swaggerhub.com/ASCOMInitiative/api/v1/switch/0/setswitchvalue
‌

Am I missing something here or is this a bug in URL assembling?

‌

LOG (Line 1-57 show the correct device connection and parameter queries, line 58 has the setswitchvalue request URL in question):
```
GET /api/v1/switch/0/interfaceversion?ClientTransactionID=1377&ClientID=11674 HTTP/1.1
REPLY: {'Value': '0001', 'ErrorNumber': 0, 'ErrorMessage': ''}

PUT /api/v1/switch/0/connected HTTP/1.1
REPLY: {'Value': '', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/maxswitch?ClientTransactionID=1379&ClientID=11674 HTTP/1.1
REPLY: {'Value': 1, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/canwrite?ClientTransactionID=1380&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 1, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchname?ClientTransactionID=1381&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 'TESTSWITCH', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchdescription?ClientTransactionID=1382&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 'SWITCHDRIVERTEST', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchvalue?ClientTransactionID=1383&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 0.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/maxswitchvalue?ClientTransactionID=1384&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 2.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/minswitchvalue?ClientTransactionID=1385&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 0.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/switchstep?ClientTransactionID=1386&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 1.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/description?ClientTransactionID=1387&ClientID=11674 HTTP/1.1
REPLY: {'Value': 'SWITCHDRIVERTEST', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/driverinfo?ClientTransactionID=1388&ClientID=11674 HTTP/1.1
REPLY: {'Value': 'SWITCHDRIVERTEST', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/driverversion?ClientTransactionID=1389&ClientID=11674 HTTP/1.1
REPLY: {'Value': '0001', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/driverversion?ClientTransactionID=1390&ClientID=11674 HTTP/1.1
REPLY: {'Value': '0001', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchname?ClientTransactionID=1391&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 'TESTSWITCH', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchvalue?ClientTransactionID=1392&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 0.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchname?ClientTransactionID=1393&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 'IMP85', 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchvalue?ClientTransactionID=1394&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 0.0, 'ErrorNumber': 0, 'ErrorMessage': ''}

GET /api/v1/switch/0/getswitchname?ClientTransactionID=1398&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 'IMP85', 'ErrorNumber': 0, 'ErrorMessage': ''}

PUT /api/v1/switch/0/setswitchvalue HTTP/1.1
REPLY: {'Value': '', 'ErrorNumber': 1025, 'ErrorMessage': 'Invalid value'}

GET /api/v1/switch/0/getswitchvalue?ClientTransactionID=1399&ClientID=11674&ID=0 HTTP/1.1
REPLY: {'Value': 0.0, 'ErrorNumber': 0, 'ErrorMessage': ''}
```
‌

astroman133, 2021-05-30:
I am just learning about Alpaca and don't look at Alpaca issues very often. Sorry that your issue has generated any comments until now. From the Alpaca API documentation:
For GET operations, parameters should be placed in the URL query string and for PUT operations they should be placed in the body of the message.
The SetSwitch value is a PUT operation, so the id, value, ClientID, and ClientTransactionID values should go in the body of the request, not in the query string.
Hopefully, Peter or Daniel will add to this if my info is incorrect or imcomplete.

Peter-Simpson, 2021-06-02:
Hi Roman,

I tried out your CURL request: 

`curl -X PUT "https://virtserver.swaggerhub.com/ASCOMInitiative/api/v1/switch/0/setswitchvalue" -H "accept: application/json" -H "Content-Type: application/x-www-form-urlencoded" -d "Id=&Value=2.0&ClientID=0&ClientTransactionID=1234"`

and value for Id is missing, the CURL string should include something like Id=0. It worked fine here when I made this change and tested it against an ASCOM Remote Server instance.

Sadly, SmartBear have a bug in their system that currently prevents our virtserver.swaggerhub.com instance from  working and always returns a 500 HTTP error.

Your best bet for debugging, if you have access to a WIndows PC, is to install the ASCOM Remote Server and configure it to serve the Platform's Switch simulator.

Sorry for the long delay in replying

Best wishes, Peter

closed 2021-06-02 by Peter-Simpson