ASCOMInitiative/ASCOMRemote #34: Extra listening IP adres on ascom remote

calberts, 2021-11-10:
Hoi

I have a question although I think it's not a bug but Request for enhancement

I use Alpaca for my remote observatory, I  love it. I use it to open my roof and query my PC for the status of the telescope with ascom remote. That all is on the local network with the observatory. To control and access the observatory I make use of zero-tier (https://www.zerotier.com/). Which introduced an overlay network with another network IP space.
ALPACA Remote Servers is only able to listen on one network one 1 address, 
My enhancement question is to add an extra one listing IP address in Ascom remote for the zerotier network.

This has added value for all remote observatory's 
The reason for me is that I like to use the sky safari from my Mac / iPad at home which has full alpaca support presently added

In this way, we control equipment at home but still also local in the observatory

Looking forward to this enhancement.

With regards 

Chris Alberts

Peter-Simpson, 2021-11-11:
Hi Chris,

The latest ASCOM remote release [here](https://github.com/ASCOMInitiative/ASCOMRemote/releases) has an "All IP Addresses" option, does this address your need?

Best wishes, Peter

[image]

calberts, 2021-11-11:
Peter

I will check this, I'm still on version 7500 fist update to the last release ;-)

Thanks for you reaction

Chris

calberts, 2021-11-12:
Peter

this did the Job, thanks ist working

Chris

closed 2021-11-12 by calberts

Peter-Simpson, 2021-11-12:
Hi Chris,

That's great, I'm glad it's working OK.

Best wishes, Peter