ASCOMInitiative/ASCOMRemote #35: This is not an issue, but improvment

StefanoTesla, 2021-12-07:
I hope one day, to see ascom alpacha in a lot of devices, cause I love it.

But...

Ascom standard is a device aggregator, and with alpacha finally we talk about api comunication, and it's fantastic.

The main problem is that are one side comunication, it should be very usefull if my device can ask to ascom server if another device is connected and read(only) it's state.

Example:

My dome is open and my telescope is slewing, I close the dome for mistake, and I will have a crash...

If the dome can ask to ascom server if a scope is connected and it's state i can prevent a crash..

What you think?

Peter-Simpson, 2022-01-10:
Hi Stefano,

I've assessed the effort to implement your suggestion and it turns out to be high because the Remote Server can be configured to access COM drivers directly or when isolated and running in separate threads. and is architected to deal with IP based queries to devices rather than internal calls from itself.

Why don't you query the telescope connected state directly yourself? Surely this would be the easiest solution?

Best wishes, Peter

StefanoTesla, 2022-01-10:
Oh yes for sure is not just a byte more on the Ascom platform, it's an hard implementaton.

In my case my dome board is made by an Esp32, but the telescopi is via usb and I can't query him cause is a standard Ascom via usb, I should create a software for get the information via Ascom driver and create a Virtual server to query with ESP32... This can be an easiest way to get the information effectively...