ASCOMInitiative/ASCOMRemote #37: Clarification on Connecting Remotely

michealroberts, 2022-08-15:
So, I have a Alpaca ASCOM API remote server set up on a VM machine on Azure*, at the port 8010 (***_for testing purposes_**)

However, for some reason, the connection hangs and is unable to respond.

Within the Windows Server VM, if I open any browser, and say navigate to: `http://127.0.0.1:8010/api/v1/telescope/0/altitude` I see the following response:

```
{"Value":38.92139,"ClientTransactionID":0,"ServerTransactionID":11,"ErrorNumber":0,"ErrorMessage":"","DriverException":null}
```

So I can confirm that it seems to be working, and working as intended.

I have also whitelisted inbound connections to 8010 within the VM, using Windows Firewall rules, and I have also added an inbound port rule to the network handling, but there still is the infinite hang.

There is certainly something I have done wrong in configuring the server, perhaps something to do with this:

[image]

However, when I add the IP address of the public IP address for the VM, I see the following issue:

[image]

So, it can't seem to discover ... 

Would anyone be able to work with me to understand how to setup on a VM?

michealroberts, 2022-08-15:
Ooof, always the way, as soon as you try something it works! Haha

So it seems that if you set it to an IP address you need to stop and the start the server again... D'oh!

closed 2022-08-15 by michealroberts

Peter-Simpson, 2022-08-15:
Hi Michael, glad you have it sorted!

Best wishes, Peter