ASCOMInitiative/ASCOMRemote #42: Cannot Install ASCOMRemote

ispybadguys, 2022-12-04:
I have a machine with an up-to-date copy of 64bit Windows 10 and NET Framework 4.8 installed. Launching ASCOMRemote.6.6.8324.21963.setup gives me a error "This App cannot run on your PC". What am I doing wrong?

Peter-Simpson, 2022-12-31:
Hi,

Sorry for the delay in replying. Is this still an issue? 

The setup utility's error messages are quite specific about the nature of any issues that it finds and doesn't contain the text "This App cannot run on your PC" that you are seeing.

Please can you post a screen-shot of the dialogue; it may contain a clue as to where the message originates.

Best wishes, Peter

ispybadguys, 2023-01-01:
Peter

I will check tomorrow but i am pretty sure this message is a Windows message saying the setup app cannot run.

Kurt

⁣Get BlueMail for Android ​

Peter-Simpson, 2023-01-01:
Hi Kurt,

 

If this is still an issue, please can you also run and attach a complete Diagnostics log to your next post as described here: https://ascomtalk.groups.io/g/Help/wiki/18368 

 

Best wishes, Peter

ispybadguys, 2023-01-02:
Peter

I downloaded ASCOMRemote.6.6.8324.21963.setup.exe from github. When I run the exe on a Windows 10 machine I get the popup

It is almost like it is a 32bit executable.

Kurt

Peter-Simpson, 2023-01-09:
Hi Kurt, 

ASCOM Remote is a 32bit executable and will present an error dialogue if you tell it to load a 64bit driver.  I can't help further without a screen shot of the error dialogue and the ASCOM Diagnostics log that I requested in my previous posts.

Best wishes, Peter

ispybadguys, 2023-01-09:
Thanks Peter

I wasn’t trying to be difficult. I didn’t send the error log because the ASCOM Remote wouldn’t install so I thought there wouldn’t be anything in terms of a log. Is there  64bit version of ASCOM Remote?

Thanks

Kurt

> On Jan 9, 2023, at 12:34 AM, Peter Simpson ***@***.***> wrote:
> 
> present

Peter-Simpson, 2023-01-09:
Hi Kurt,

 

No there isn’t a 64bit version at the moment, this is to ensure that 32bit “in-process” COM drivers run OK. Are you trying to run a 64bit only driver? If so, which one?

 

Best wishes, Peter

ispybadguys, 2023-01-09:
Peter

I have a 64bit Windows 10 system in my observatory. I run NINA and I am trying to switch the ROR controller out from a ASCOM system to an Alpaca system. NINA doesn’t let me use the Alpaca interface unless I run Remote and setup the device. So that is my interest.

Kurt

Peter-Simpson, 2023-01-09:
Hi Kurt,

 

Sorry if I’m being dense but I’m perplexed by the description of what you are trying to achieve. I understand that NINA is the client but don’t understand what you mean here:

 

“I am trying to switch the ROR controller out from a ASCOM system to an Alpaca system”

 

ASCOM Remote presents COM clients as Alpaca devices but I thought that NINA couldn’t communicate with Alpaca devices at the moment so I’m not sure how ASCOM Remote will help. 

 

Please can you explain more about what you had that was working and how you want that to change?

 

Many thanks, Peter

Peter-Simpson, 2023-01-09:
Hi Kurt,

 

I have to confess I don’t know anything about NINA so this may not be a conversation that’s going to answer your questions! 😁

 

I wonder whether you will get further by initially asking questions on the NINA Discord server: https://discord.com/invite/rWRbVbw and then coming back to me if you have specific ASCOM related questions. These guys will know how to get the best out of NINA and will almost certainly have already found solutions to getting NINA and ASCOM to integrate effectively.

 

It’s clear that NINA is the controlling client application, and it sounds like you currently have a traditional ASCOM COM driver for your ROR. Who wrote that driver, was it provided by the ROR control system manufacturer, or did you write that yourself? NINA should be able to control the ROR through the existing COM driver – so long as they both reside on the same PC.

 

If you want to use two PCs then, yes, Alpaca / ASCOM Remote is a potential solution. In this configuration:

*	PC1

*	Runs NINA
*	Uses an Alpaca Dynamic Client that is created by the Chooser to talk to either

*	ASCOM Remote running on PC2 or
*	An independent Alpaca device 

*	PC2

*	Runs ASCOM Remote which

*	Presents a 32bit COM ROR driver directly or
*	Presents the Platform Device Hub, which can host a 64bit COM ROR driver

 

Is this what you are looking to achieve or is it something else?

 

Best wishes, Peter

Peter-Simpson, 2024-09-26:
Closing due to no further activity.

closed 2024-09-26 by Peter-Simpson