ASCOMInitiative/ASCOMRemote #47: Connection of 2 same focuser

CTugTEZ, 2023-02-01:
Hi,

I am trying to connect 2 FLI Atlas Focuser to ASCOM Remote Server via their USB 2.0 connections. Both focuser are working properly with their application of FLIFocus. Connection dialog recognize them with their driver flifoc0 and flifoc1. ASCOM also properly recognize both devices. 
[image]

But when I try to define them in remote server configuration, i.e. I am making configuration for Device Number: 0, I select "Setup" and choose flifoc0. Making same procedure for Device Number: 1. When I selected the flifoc1 for Device, it is also changing flifoc0 to flifoc1. 

[image]

I am developing a connection code for both. It is connecting just 1 device which I selected at the end in remote server configuration.
```
        F1 = Focuser('xx.zzz.kkk.ll:mm', 0)
        F2 = Focuser('yy.aaa.bbb.cc:dd', 1)
```

Thank for the suggestions in advance!
Best!
Cihan

Peter-Simpson, 2023-02-19:
Hi Cihan,

This sounds like it could be an issue with the way that FLI have implemented their driver or it may be a Remote Server issue because FLI have used the same description text for both devices.

First off, let's try to see if this is something just associated with the remote server or whether it is associated with all applications. To do this I suggest using the Diagnostics Chooser application that is supplied with the Platform as described [here](https://ascomtalk.groups.io/g/Help/wiki/29544). Do you see the same or different behaviour when selecting your two focusers?

If the problem is the same then this probably needs to be referred to FLI, if the problem is only in ASCOM Remote then please will you create a full Diagnostics listing as described [here](https://ascomtalk.groups.io/g/Help/wiki/18368) and attach the full diagnostics txt file to your next post.

Best wishes, Peter

Peter-Simpson, 2024-09-26:
Closing due to no further activity.