ASCOMInitiative/ASCOMRemote #53: Process terminated - Unhandled exception

ppp-one, 2024-01-09:
Hi - we are using ASCOM Remote with an Andor iKon-L camera (32 bit driver), and ASCOM Remote (sometimes, but often) crashes after a startexposure is called, specifically after a sub-frame of 64x64 pixels is made by setting (in the following order) NumX -> 64, NumY -> 64, StartX -> 1012, StartY -> 992.  It behaves very well otherwise 👍 . See the following screenshots of the Windows event log and Remote's log.

[image]

[image]

From the camera driver, we get the following Exception: 'Setting shutter failed returned error code 20072'.  We have a 1s sleep before calling the a start exposure after the sub frame is set, but maybe this is not enough.

I will run more tests in the coming days and similarly contact Andor. Any ideas on how to resolve/debug welcomed!

Thank you,
Peter

Peter-Simpson, 2024-01-10:
Hi Peter,

ASCOM Remote has a logger for unhandled exceptions so you should find a log file named ASCOM.RemoteAccessServerException.xxx in your Documents\ASCOM\Logs 2023.XX.YY folder where XX and YY are the month and day on which the issue occurred.

Can you have a look to see if you can find and post any of these logs?

Best wishes, Peter

ppp-one, 2024-01-11:
Hi Peter, thank you for the reply - no luck finding the logs sadly.  I guess it crashed before it could write one.

Peter-Simpson, 2024-01-11:
Hi Peter, that's surprising I thought that handler of last resort was always called by Windows when an unhandled exception occurred. 

The exception in the event log entry you posted shows the originating exception site as being in the namespace ASCOM.SDK2.Camera, although this namespace is not part of the Remote Server code base.

I presume that the namespace is part of an in-process (DLL) camera driver that runs within the context of the Remote Server application. Any significant failure within the driver will appear as a failure of the Remote Server, although the Remote Server is really a victim rather than a cause of the issue.

I suggest reporting the namespace to the driver author and hopefully they will recognise it as part of their driver.

Best wishes, Peter

Peter-Simpson, 2024-09-26:
Closing due to no activity for 9 months.

closed 2024-09-26 by Peter-Simpson