ASCOMInitiative/ASCOMRemote #54: Crash when entering setup

sbraybrooke63, 2024-04-14:
Hi Peter. I have installed Ascom Remote Server V6.7.1 build 4785. It loads an old config which used CPWI which I no longer use. I enter the setup page to remove it, but the setup screen disappears after 1 second and the application crashes. Any ideas please? Windows 11 Pro, V 23H2, 64-bit operating system, x64-based processor. 
[image]
[image]

Peter-Simpson, 2024-04-14:
Hi,

I plan to add a reset button to the GUI, you are the second person who has found themselves in this position. For now, I've just added a Wiki entry describing how to delete the current configuration here: [Reset ASCOM Remote](https://ascomtalk.groups.io/g/Help/wiki/36534).

Please let me know if this gets you going again.

Best wishes, Peter

sbraybrooke63, 2024-04-14:
Thank you very much Peter. The wiki link doesn't seem to accept my login for some reason. But I have found a work around; I  uninstalled CPWI and reinstalled ASCOM remote. It then reported an error in the setup but it let me edit the setup so I could add in my new mount (NYX-101). I now appear to have comms established between my "indoors Mac" and my "outdoors telescope PC" albeit with some oddities which I am now investigating (probably in other apps). Many thanks!

Peter-Simpson, 2024-04-14:
Great, thanks for letting me know, I'm not sure about the source of your Wiki link issue, the page opens OK here.

Now I'm off to add a reset button to the ASCOM Remote Gui!

Best wishes, Peter

Peter-Simpson, 2024-04-15:
A clear way to reset the configuration has been added to the ASCOM Remote GUI so I'm closing this issue. The update will be part of the Platform 7 release.

Best wishes, Peter

sbraybrooke63, 2024-04-15:
Amazing Peter - thank you!