ASCOMInitiative/ASCOMRemote #56: GET all current device properties with one call?

chvvkumar, 2024-08-03:
**Synopsys:**
I am trying to write a home assistant plugin that will show the current system status on a dashboard. 

**Process:**
I am using Home Assistant's [RESTful Sensor](https://www.home-assistant.io/integrations/sensor.rest/) to GET API endpoints individually and parse the JSON response for specific values like so:

Example from `/api/v1/telescope/0/declination`
[image]

Currently, I am using these endpoints:

```
FlatPanel
/api/v1/covercalibrator/0/connected
/api/v1/covercalibrator/0/calibratorstate

Camera
/api/v1/camera/0/connected
/api/v1/camera/0/ccdtemperature
/api/v1/camera/0/cooleron
/api/v1/camera/0/coolerpower
/api/v1/camera/0/percentcompleted
/api/v1/camera/0/percentcompleted
/api/v1/camera/0/subexposureduration
/api/v1/camera/0/imagearray

FiltweWheel
/api/v1/filterwheel/0/connected
/api/v1/filterwheel/0/position

Focuser
/api/v1/focuser/0/connected
/api/v1/focuser/0/temperature

Telescope
/api/v1/telescope/0/connected
/api/v1/telescope/0/athome
/api/v1/telescope/0/atpark
/api/v1/telescope/0/declination
/api/v1/telescope/0/rightascension
/api/v1/telescope/0/sideofpier
```

**Issue:**
The way the API currently works, I am creating a single sensor for each API endpoint I want to parse. This causes multiple API calls to ASCOMRemote to get one set of metrics. 

**Inquiry:**
I have looked through the [Swagger documentation](https://www.ascom-standards.org/api/) but I was not able to find a single endpoint where I can GET all the status as a JSON object in one go with one API call. Does such a thing exist already or is there a way to get this information like the example I am thinking below?

```
"root":{
    "covercalibrator": {
        "0": {
            "connected": "",
            "brightness": "",
            "x": "",
            "y": "",
            "z": ""
        }
    },
    "telescope": {
        "0": {
            "connected": "",
            "alignmentmode": "",
            "x": "",
            "y": "",
            "z": ""
        }
    }
.
.
.

}
```

Peter-Simpson, 2024-08-04:
Hi Kumar,

Our Platform 7](https://github.com/ASCOMInitiative/ASCOMPlatform/releases) update (currently at release candidate 4) introduces support for a new common property: [DeviceState](https://ascomtalk.groups.io/g/Developer/message/7066). This will help you by returning all the operational properties of a device in one call e.g. all the Camera properties you mention or all the Telescope properties.

At present ASCOM Remote supports the new interface versions so long as the drivers implement them, but it does not emulate the new interfaces when drivers only support Platform 6 interfaces.

Connected is not included as an operational property because we assume that every client application already knows whether or not it has connected to the device.

I can have a look at adding this but it may prove complicated to ensure 100% compatibility.

Best wishes, Peter

chvvkumar, 2024-08-06:
> Hi Kumar,
>
> Our Platform 7](https://github.com/ASCOMInitiative/ASCOMPlatform/releases) update (currently at release candidate 4) introduces support for a new common property: [...]

Thank you for your reply! 

I think I can handle device status (or other, device specific parameters) on the client side. I was just looking to see if I can get all device parameters that are already available at separate endpoints at one API endpoint. For example, this could be something like "/api/v1/status" and that would have a packaged up JSON of all the information already available at various device trees.

Peter-Simpson, 2024-08-06:
Hi Kumar,

I see what you are asking for, such a call would be an ASCOM Remote specific enhancement rather than a common feature for all devices. It will require extensive development even for devices that support DeviceState, extending it to emulate DeviceState for Platform 6 interface devices would add further complexity and development effort .

I don't see a substantial demand from the community for this feature, but you can create something similar in your client by using the Alpaca [ConfiguredDevices ](https://ascom-standards.org/api/?urls.primaryName=ASCOM%20Alpaca%20Management%20API#/Management%20Interface%20(JSON)/get_management_v1_configureddevices) management API call to get the list of available devices and then querying each device for the information.

I appreciate this still imposes network overhead but is the best work-round I can offer.

Best wishes, Peter

chvvkumar, 2024-08-08:
> Hi Kumar,
>
> I see what you are asking for, such a call would be an ASCOM Remote specific enhancement rather than a common feature for all devices. It will require extensive development even for [...]

Thank you Peter for taking the time to respond. I will explore the management API and see what I can come up with. I understand my use case is pretty narrow so I do appreciate the scope of work needed might not be worth the effort.

Thank you!

closed 2024-08-08 by chvvkumar