ASCOMInitiative/ASCOMRemote #66: [REQUEST] feat(camera): add StoreImage() route

michealroberts, 2024-09-29:
Something that I feel would be extremely beneficial is the ability to send a command to store the latest image, if any, directly from the camera as either the raw data, a FITs, an astroTIFF or XISF (or other preferred standard) on the user's local machine.

I'm happy to propose an API and prepare an RFC document for this, but first, I wondered what the appetite would be for such a command?

A use case for this would be to allow user to handle post-imaging steps. For example, storing files on the local system could work in tandem with other services on a user's local machine that could watch the filesystem for changes/additions and perform any additional pre-processing, post-processing and data analysis as a separate process.

I'm not a C# developer, so if any of this is not possible please interject: but I for-see it as working as a simple non-blocking async call that returns as soon as the command to store has been made, with a separate coroutine dispatched to store the raw image byte data.

I think this would be very advantageous for, e.g., situational awareness applications where data acquisition, analysis and storage need to be as performant as possible.

Peter-Simpson, 2024-09-30:
Hi Michael,

ASCOM Remote is a piece of "infrastructure" whose function is to translate the Alpaca protocol to COM and vice versa. I don't think adding device specific application functionality, in the way you are suggesting, is in line with its purpose.

If you wish to store images on the PC to which the camera is attached, why not install appropriate client software there and use remote desktop to view / transfer images?

Nest wishes, Peter

pedroosorio, 2026-03-03:
Hi @Peter-Simpson !

Documentation says: "Returning an image from an Alpaca device as a JSON array is very inefficient and can result in delays of 30 or more seconds while client and device process and send the huge JSON string over the network. A new, much faster mechanic called ImageBytes - [Alpaca ImageBytes Concepts and Implementation](https://www.ascom-standards.org/Developer/AlpacaImageBytes.pdf) has been developed that sends data as a binary byte stream and can offer a 10 to 20 fold reduction in transfer time. It is strongly recommended that Alpaca Cameras implement the ImageBytes mechanic as well as the JSON mechanic."

For a camera (ie SBIG ALUMA AC4040) that does not have native Alpaca interface, where I'm using the remote server to interface with it, all i can do is call imagearray right ? Is it possible to implement a method to store the image in a filesystem local to the host running the server, even if it is a non-standard extension to the server ?

My use-case doesn't really involve the use of GUIs in camera control or image acquisition.

Any help or advice is much appreciated !

michealroberts, 2026-03-04:
I wonder … there does seem to be a use case which is end users may need performant streaming of images.

I wonder if we can use something like rtc or blobs or similar @Peter-Simpson to get around latency issues …

A whole other can of worms, but I wonder if ASCOM could ever be made native cross-platform … out of the box. And not reliant on the COM protocol on Linux / Unix / Mac.

Peter-Simpson, 2026-03-05:
@michealroberts ImageBytes is already performant as a network file transfer protocol and minimises the amount of data transferred over the wire by using the smallest datatype required to hold the data, and then converting to the required data type at the destination. 

Alpaca is our way forward for cross Platform use because it uses HTTP/REST as its interface definition rather than COM, which is, as you say, essentially limited to Windows. When you say "ASCOM could ever be made native cross-platform" are you referring to the ASCOM Platform? The Platform essentially has two components:

- The ASCOM Profile, which enables ASCOM COM clients to find COM drivers and provides a configuration store
- Support Utilities that make it easier for developers to create ASCOM clients and drivers

We have created an independent set of cross-platform Utilities (Linux x64, ARM32, ARM64 and OSX) in the form of the [ASCOM Library](https://ascom-standards.org/library/html/3e0a4970-b422-4427-bb1e-789bb79e1b4f.htm) so neither of the Platform features are required by Alpaca devices, hence there is no need for a Platform on non-Windows OSs.