ASCOMInitiative/ASCOMRemote #71: Remote Server Crashes upon imagearray request

artyom-beilis, 2025-07-29:
- ASCOM platform 7 update 2 7.0.2.4675
- ASCOM Remote 6.7.1 build 4785
- camera asi zwo 224mc
- zwo ascom 6.5.30
- zwo drivers 3.25

Upon request of imagearray Remote Server crashes

```
Device Camera 0 (ASCOM.ASICamera2_2.Camera) connected OK and is available
ERROR - Device Telescope 0 (ASCOM.SynScanMobile.Telescope) failed to connect and is NOT available: No device found
ASCOM Remote Alpaca device listening URI: http://+:11111/api/
Listening for IPv4 discovery broadcasts on port 32227.
IPv6 discovery not enabled.
Server started successfully.
Device Camera 0 (ASCOM.ASICamera2_2.Camera) connected OK and is available
ERROR - Device Telescope 0 (ASCOM.SynScanMobile.Telescope) failed to connect and is NOT available: No device found
ASCOM Remote Alpaca device listening URI: http://+:11111/api/
Listening for IPv4 discovery broadcasts on port 32227.
IPv6 discovery not enabled.
Server started successfully.
GET /management/v1/configureddevices
  OK - JSON: {"Value":[{"DeviceName":"ASI Camera (2)","DeviceType":"Camera","DeviceNumber":0,"UniqueID":"F1E4A97D-8380-4715-91E8-0F5642FF42DE"},{"DeviceName":"SynScan App Driver","DeviceType":"Telescope","DeviceNumber":0,"UniqueID":"250C3D32-30F3-461A-819C-35EB9BEDCE95"}],"ClientTransactionID":0,"ServerTransactionID":1,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/connected
  OK - JSON: {"Value":true,"ClientTransactionID":0,"ServerTransactionID":2,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/gains
  DEVICE ERROR - ClientId: 0, ClientTxnID: 0, ServerTxnID: 3 - Property  Unknown is not implemented in this driver.
GET /api/v1/camera/0/gainmin
  OK - JSON: {"Value":0,"ClientTransactionID":0,"ServerTransactionID":4,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/ccdtemperature
  OK - JSON: {"Value":32.3,"ClientTransactionID":0,"ServerTransactionID":5,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/cansetccdtemperature
  OK - JSON: {"Value":false,"ClientTransactionID":0,"ServerTransactionID":6,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/cooleron
  OK - JSON: {"Value":false,"ClientTransactionID":0,"ServerTransactionID":7,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/coolerpower
  OK - JSON: {"Value":0.0,"ClientTransactionID":0,"ServerTransactionID":8,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/exposuremin
  OK - JSON: {"Value":0.0,"ClientTransactionID":0,"ServerTransactionID":9,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/exposuremax
  OK - JSON: {"Value":2000.0,"ClientTransactionID":0,"ServerTransactionID":10,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/exposureresolution
  OK - JSON: {"Value":1E-06,"ClientTransactionID":0,"ServerTransactionID":11,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/gain
  OK - JSON: {"Value":0,"ClientTransactionID":0,"ServerTransactionID":12,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/gainmin
  OK - JSON: {"Value":0,"ClientTransactionID":0,"ServerTransactionID":13,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/gainmax
  OK - JSON: {"Value":300,"ClientTransactionID":0,"ServerTransactionID":14,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/ccdtemperature
  OK - JSON: {"Value":32.3,"ClientTransactionID":0,"ServerTransactionID":15,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/cooleron
  OK - JSON: {"Value":false,"ClientTransactionID":0,"ServerTransactionID":16,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/coolerpower
  OK - JSON: {"Value":0.0,"ClientTransactionID":0,"ServerTransactionID":17,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/sensortype
  OK - JSON: {"Value":2,"ClientTransactionID":0,"ServerTransactionID":18,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/cameraxsize
  OK - JSON: {"Value":1304,"ClientTransactionID":0,"ServerTransactionID":19,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/cameraysize
  OK - JSON: {"Value":976,"ClientTransactionID":0,"ServerTransactionID":20,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/maxbinx
  OK - JSON: {"Value":2,"ClientTransactionID":0,"ServerTransactionID":21,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/maxbiny
  OK - JSON: {"Value":2,"ClientTransactionID":0,"ServerTransactionID":22,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/exposuremin
  OK - JSON: {"Value":0.0,"ClientTransactionID":0,"ServerTransactionID":23,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/exposuremax
  OK - JSON: {"Value":2000.0,"ClientTransactionID":0,"ServerTransactionID":24,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/canabortexposure
  OK - JSON: {"Value":true,"ClientTransactionID":0,"ServerTransactionID":25,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/bayeroffsetx
  OK - JSON: {"Value":0,"ClientTransactionID":0,"ServerTransactionID":26,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/bayeroffsety
  OK - JSON: {"Value":0,"ClientTransactionID":0,"ServerTransactionID":27,"ErrorNumber":0,"ErrorMessage":""}
PUT /api/v1/camera/0/binx
  OK - JSON: {"ClientTransactionID":1,"ServerTransactionID":28,"ErrorNumber":0,"ErrorMessage":""}
PUT /api/v1/camera/0/biny
  OK - JSON: {"ClientTransactionID":2,"ServerTransactionID":29,"ErrorNumber":0,"ErrorMessage":""}
PUT /api/v1/camera/0/numx
  OK - JSON: {"ClientTransactionID":3,"ServerTransactionID":30,"ErrorNumber":0,"ErrorMessage":""}
PUT /api/v1/camera/0/numy
  OK - JSON: {"ClientTransactionID":4,"ServerTransactionID":31,"ErrorNumber":0,"ErrorMessage":""}
PUT /api/v1/camera/0/startexposure
  OK - JSON: {"ClientTransactionID":5,"ServerTransactionID":32,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/imageready
  OK - JSON: {"Value":false,"ClientTransactionID":0,"ServerTransactionID":33,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/imageready
  OK - JSON: {"Value":false,"ClientTransactionID":0,"ServerTransactionID":34,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/imageready
  OK - JSON: {"Value":true,"ClientTransactionID":0,"ServerTransactionID":35,"ErrorNumber":0,"ErrorMessage":""}
GET /api/v1/camera/0/imagearray
```

artyom-beilis, 2025-07-29:
Version 6.6 works

artyom-beilis, 2025-07-30:
Ok attaching crash logs (with both asi camera and simulator camera) 

The last relevant lines

```
18:59:23.945        0      0      0    APIRequestCallback     Thread 7 - Request received. Is thread pool thread: True. Is background thread: True.
18:59:23.945        0      0      0    APIRequestCallback     Thread 7 - Setting up new call back to wait for next request 
18:59:23.945        0      0      0    APIRequestCallback     The configured server IP address is + or * so we will process this request.
18:59:23.945        0      0      0    APIRequestCallback     Thread 7 - Processing received message.
18:59:23.945        0      0     29    RequestReceived        GET URL: /api/v1/camera/0/imageready, Protocol: HTTP/1.1, Thread: 7
18:59:23.946        0      0     29    RequestReceived        Header Connection = close
18:59:23.946        0      0     29    RequestReceived        Header Accept = */*
18:59:23.946        0      0     29    RequestReceived        Header Accept-Encoding = 
18:59:23.946        0      0     29    RequestReceived        Header Host = 172.25.192.1:11111
18:59:23.946        0      0     29    RequestReceived        Header User-Agent = cpp-httplib_for_ols/0.23.1
18:59:23.946        0      0     29    imageready             Device only supports serialised access - command lock acquired for device camera/0 on REST thread 7
18:59:23.946        0      0     29    imageready             ProcessRequestAsync - Sending command to camera/0 on REST thread 7
18:59:23.947        0      0     29    imageready             DriverCommand has received a command for camera/0 on FORM thread 19 Apartment state: STA Is background: True Is thread pool thread: False
18:59:23.947        0      0     29    imageready             Processing driver command on WORKER thread 46
18:59:23.947        0      0     29    imageready             DriverCommand has started the command for camera/0 on FORM thread 19
18:59:23.947        0      0     29    imageready             OK - no exception. Thread: 46, JSON: {"Value":true,"ClientTransactionID":0,"ServerTransactionID":29,"ErrorNumber":0,"ErrorMessage":""}
18:59:23.949        0      0     29    imageready             Completed Encoding.GetBytes, array length: 97
18:59:23.949        0      0     29    imageready             Before setting response bytes - Length: 97, Response is null: False
18:59:23.949        0      0     29    imageready             Before writing 97 bytes to output stream
18:59:23.949        0      0     29    imageready             After writing bytes to output stream
18:59:23.949        0      0     29    imageready             After closing output stream
18:59:23.949        0      0     29    imageready             ProcessRequestAsync - Command completed for camera/0 using WORKER thread 46 on REST thread 7
18:59:23.949        0      0     29    imageready             Device only supports serialised access - command lock released for device camera/0 from REST thread 7
18:59:23.949                           
18:59:23.950        0      0      0    APIRequestCallback     Thread 7 - Request received. Is thread pool thread: True. Is background thread: True.
18:59:23.950        0      0      0    APIRequestCallback     Thread 7 - Setting up new call back to wait for next request 
18:59:23.950        0      0      0    APIRequestCallback     The configured server IP address is + or * so we will process this request.
18:59:23.950        0      0      0    APIRequestCallback     Thread 7 - Processing received message.
18:59:23.950        0      0     30    RequestReceived        GET URL: /api/v1/camera/0/imagearray, Protocol: HTTP/1.1, Thread: 7
18:59:23.953        0      0     30    RequestReceived        Header Connection = close
18:59:23.953        0      0     30    RequestReceived        Header Accept = application/imagebytes
18:59:23.953        0      0     30    RequestReceived        Header Accept-Encoding = 
18:59:23.953        0      0     30    RequestReceived        Header Host = 172.25.192.1:11111
18:59:23.953        0      0     30    RequestReceived        Header User-Agent = cpp-httplib_for_ols/0.23.1
18:59:23.953        0      0     30    imagearray             Device only supports serialised access - command lock acquired for device camera/0 on REST thread 7
18:59:23.953        0      0     30    imagearray             ProcessRequestAsync - Sending command to camera/0 on REST thread 7
18:59:23.953        0      0     30    imagearray             DriverCommand has received a command for camera/0 on FORM thread 19 Apartment state: STA Is background: True Is thread pool thread: False
18:59:23.953        0      0     30    imagearray             DriverCommand has started the command for camera/0 on FORM thread 19
18:59:23.953        0      0     30    imagearray             Processing driver command on WORKER thread 47
```

Tell me if more information is needed

[Logs 2025-07-30.zip](https://github.com/user-attachments/files/21516334/Logs.2025-07-30.zip)

Peter-Simpson, 2025-07-31:
This is very odd, ASCOM Remote has been stable for many years now and I've never seen it simply stop at the place indicated in the logs.

Please can you:
- install this latest [Platform Release Candidate](https://download.ascom-standards.org/beta/ASCOMPlatform703-rc.8.4764.exe). This will fix any corruption that may have happened to your current Platform and ASCOM Remote installation.
- Switch back to the original Platform simulators by running the Diagnostics tool and unchecking the Options / Use Omni-Simulators as Platform simulators checkbox. 
[image]

- Start ASCOM Remote,  press the Camera Device Setup button on the setup screen and set the width and height to 640 x 480.
[image]

- Close the Camera properties dialogue and the ASCOM Remote Setup windows. AR should now be running and hosting the original Platform simulator.

Does your client now work? If not, I suggest:
- Download and install our [Conform Universal](https://github.com/ASCOMInitiative/ConformU/releases/tag/v4.1.0) driver/device test program. 
- Configure the Conform Settings like this:
[image]

- Configure the Camera settings like this:
[image]

- Select the ASCOM Remote Camera device like this:
[image]

- Run a test from the Home Page
- Create a Diagnostics log as described [here](https://ascomtalk.groups.io/g/Help/wiki/18368).
- Post a zip of the logs folder as before

Hopefully this will shed some more light on the issue.

Best wishes, Peter

artyom-beilis, 2025-07-31:
> First, I suggest switching over from the OmniSims to the Platform simulators as described in my post because
> the Platform camera simulator driver runs within the ASCOM Remote process and you won't need to worry about
> what is happening inside a separate OmniSimulator process.

I'll do it and record the logs.

But also note it happens also with ASI ZWO 224MC Camera (1.2MP camera with tiny frame) and not
only simulator in same place.

Thanks,
Artyom

Peter-Simpson, 2025-07-31:
> But also note it happens also with ASI ZWO 224MC Camera

Understood, I don't have your camera, hence using the simulator enables me to investigate why this is happening. If we fix the simulator scenario, it's likely that the ASI camera will also work.

Best wishes, Peter

artyom-beilis, 2025-08-02:
I uninstalled ascom platform and installed RC you suggested
I configured it as this

[image]
[image]

It still closes upon request.

I tested with Sharpcap with Alpaca and it works but with my app closes immediately (latest log shows it well)

[Logs 2025-08-02.zip](https://github.com/user-attachments/files/21557047/Logs.2025-08-02.zip)

Here is interesting file I found:

```
05:43:16.010 Main                      Un-handled exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at ASCOM.Remote.ServerForm.ReturnImageArray(RequestData requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\ServerForm.cs:line 6432
   at ASCOM.Remote.ServerForm.ProcessDriverCommand(RequestData requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\ServerForm.cs:line 3806
   at ASCOM.Remote.DriverHostForm.ProcessCommand(Object requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\DriverHostForm.cs:line 91
   at System.Threading.Thread.StartCallback()
```

Another clue is that sharpcap sends ClientID in URL and my doesn't

This is OpenLiveStacker

```
05:43:16.003        0      0   1515                           RequestReceived        GET URL: /api/v1/camera/0/imagearray, Protocol: HTTP/1.1, Thread: 5
```

This is SharpCap

```
05:41:44.880        0      0    208                           RequestReceived        GET URL: /api/v1/camera/0/imagearray?ClientID=1&ClientTransactionID=202, Protocol: HTTP/1.1, Thread: 8
```

Now yesterday I released first Windows version as beta: https://github.com/artyom-beilis/OpenLiveStacker/releases/tag/wintest_1 so if you want to you can

If you want to try it - just unzip it, select alpaca camera, and start (the browser window will open) and select the camera and than in next screen press "Stream"

artyom-beilis, 2025-08-02:
I added clientid/transationid to queries in my app still same crash

```
07:09:33.963 Main                      Un-handled exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at ASCOM.Remote.ServerForm.ReturnImageArray(RequestData requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\ServerForm.cs:line 6432
   at ASCOM.Remote.ServerForm.ProcessDriverCommand(RequestData requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\ServerForm.cs:line 3806
   at ASCOM.Remote.DriverHostForm.ProcessCommand(Object requestData) in C:\ASCOM Build\Export\AscomRemote\Remote Server\DriverHostForm.cs:line 91
   at System.Threading.Thread.StartCallback()
```

Here the log:

```
07:09:33.954        0      0     32                           RequestReceived        GET URL: /api/v1/camera/0/imagearray?ClientID=36935&ClientTransactionID=29, Protocol: HTTP/1.1, Thread: 3
07:09:33.954        0      0     32                           Query Parameter        ClientID = 36935
07:09:33.954        0      0     32                           Query Parameter        ClientTransactionID = 29
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Header Connection = close
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Header Accept = application/imagebytes
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Header Accept-Encoding = 
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Header Host = 10.0.0.17:11111
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Header User-Agent = cpp-httplib_for_ols/0.23.1
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Parameter ClientID = 36935
07:09:33.956    36935     29     32    10.0.0.17:53444        RequestReceived        Parameter ClientTransactionID = 29
07:09:33.956    36935     29     32    10.0.0.17:53444        imagearray             Device only supports serialised access - command lock acquired for device camera/0 on REST thread 3
07:09:33.956    36935     29     32    10.0.0.17:53444        imagearray             ProcessRequestAsync - Sending command to camera/0 on REST thread 3
07:09:33.956    36935     29     32    10.0.0.17:53444        imagearray             DriverCommand has received a command for camera/0 on FORM thread 16 Apartment state: STA Is background: True Is thread pool thread: False
07:09:33.956    36935     29     32    10.0.0.17:53444        imagearray             DriverCommand has started the command for camera/0 on FORM thread 16
07:09:33.956    36935     29     32    10.0.0.17:53444        imagearray             Processing driver command on WORKER thread 34
```

artyom-beilis, 2025-08-02:
Tell me please if you need the 2nd part with full diagnostic or this trace gives you enough hints?

(Because I uninstall 6.7 and install 6.6 to continue to work all the time)

Peter-Simpson, 2025-08-02:
Hi Artyom,

I knew the "unhandled exception handler" would come in useful one day! 😁

Thanks for the stack trace, it ends at the first line of a method but doesn't pinpoint the line with the issue. I've added a lot more debug logging to the method, which will hopefully provide more evidence. I've also added a try-catch block round the whole of the method, which should keep ASCOM Remote up and running despite it having the issue.

Please can you:
- Install this latest Platform version, which contains the updated ASCOM Remote version: [Revised Platform](https://download.ascom-standards.org/beta/ASCOMPlatform710-rc.1.4768.exe)
- Ensure the ASCOM Remote debug options are enabled as shown earlier.
- Reproduce the problem
- Post the whole of the ASCOM Remote log,

Hopefully this will shine more light on why this is happening.

Best wishes, Peter

artyom-beilis, 2025-08-02:
> I've also added a try-catch block round the whole of the method, which should keep ASCOM Remote up and running despite it having the issue.

Actually it is good it crashed, otherwise I would be wondering why my SW isn't working and last thing would be checking ascom-remote `:-)`

I'll check and send updates ASAP

Peter-Simpson, 2025-08-03:
Thanks for the log, I've tracked down what I think is happening: the trigger is that the request from your client includes an empty Accept-Encoding header. ASCOM Remote was coded to expect a value if this header was present and fell over when trying to read the non-existent value.

I've added handling for empty Accept-Encoding headers in an updated Platform, please can you let me know if it now works as expected.

[Updated Platform](https://download.ascom-standards.org/beta/ASCOMPlatform710-rc.1.4768a.exe).

Best wishes, Peter

artyom-beilis, 2025-08-03:
Yes it resolved the issue!

artyom-beilis, 2025-08-03:
I also added "identity" encoding to the list in the client for imagearray request and now it works with official 6.7 https://github.com/artyom-beilis/OpenLiveStacker/commit/f0c871b5a157b543e813885ee22cd222fd674a64